ophub/luci-app-amlogic #3: Question

adisubagja, 2021-07-13:
Whats effect if failed update ? on b860

ophub, 2021-07-13:
No effect, just reinstall.

What version of the OPENWRT firmware are you now, and what version of the plug-in?
I have done installations/updates hundreds of times, but I haven't found any failures for the time being.
Before you update, upgrade the plug-in to the latest version.

adisubagja, 2021-07-13:
Im using firmware lede ophub old ver.
This have high risk to boot emmc?
i want flash latest version and next update i want using this tools
Thanks for you answer

ophub, 2021-07-13:
Install the latest version once. It will be fine later.
At the beginning of June, lede upgraded the 21.02 branch (firmware serial number 60 and above are new branches), so you may have a lot of problems when upgrading from the previous 18.06 branch.

closed 2021-07-13 by ophub