ophub/luci-app-amlogic #6: error install in emmc

radityabh, 2021-08-18:
[image]
i have this error when install to emmc, have any solution? thank you

ophub, 2021-08-18:
try again.

closed 2021-08-19 by ophub

radityabh, 2021-08-20:
> * No need to change, click to install again.
>   https://github.com/ophub/luci-app-amlogic/blob/b43202fddec3b74fee38c8a9e00c11906a9506c4/luci-app-amlogic/root/usr/bin/openwrt-install#L396
> * Where did you download the openwrt firmware and integrate the `fdisk` component when compiling?
> 
> https://github.com/ophub/amlogic-s9xxx-openwrt/blob/4d73755ee34d4ba2cd31a7056486ea40f3a2067a/router-config/lede-master/.config#L562

i fork your repo and compile it. yes i integrate the fdisk component.
[image]
i try several time, still error. i change the stb in x96max+ also the error apear.
[image]
this is my config
https://github.com/radityabh/amlogic-s9xxx-openwrt/blob/04092f04bd7bace3f8be3e8c2b992d520c63b599/router-config/lede-master/.config