ophub/luci-app-amlogic #10: 小建议

binge8, 2022-01-29:
目前在线升级好像只能设置从github的releases下载，希望开放从个人网盘或服务器下载，方便国内直连下载

ophub, 2022-01-29:
编译openwrt的源码库和第三方插件都在github上，你可以实践一下，在国内编译openwrt时可能都拉取不了源代码。国内某个类似的网站，发布固件有存储空间的容量限制，使用限制，会员限制… 现在国内没有合适的环境，可能在github上编译是目前唯一的选择，你的理想是我们一起的理想，新年伊始，愿我们都心想事成……

binge8, 2022-01-29:
在github上编译是不错的选择，不过我编译完可以打包下载到我的自建小型服务器上，完全可以实现

ophub, 2022-01-29:
代码是开源的，完全开放，支持二次开发，你可以把下载的api处改一下，在你打包的openwrt里面集成自己修改的插件。除了wget下载文件的这个功能外，其他文面都可以通用

binge8, 2022-01-29:
首先我是小白，再开发能力有限，再者，重复造轮子也没那个必要。只要在你的基础上，加个小功能，能直接自己填写想要的下载链接，就完美了。当然，也许你有你的考虑，我只是个建议，祝新年快乐

ophub, 2022-01-29:
你下载github的这个文件看一眼就明白了，可以什么都不改，只把api文件里面的内容改成你的文件的地址。这样你在插件里写自己的网站即可

binge8, 2022-01-29:
好的  明白  谢谢

closed 2022-01-29 by binge8

ophub, 2022-01-29:
```
    "node_id": "RE_kwDOFIyF_M4DI65_",
    "tag_name": "flippy_openwrt",
    "target_commitish": "main",
    "name": "v66",
    "draft": false,
    "prerelease": false,
    "created_at": "2021-11-03T03:02:57Z",
    "published_at": "2021-11-04T09:36:33Z",
    "assets": [
      {
        "url": "https://api.github.com/repos/breakings/OpenWrt/releases/assets/48605524",
        "id": 48605524,
        "node_id": "RA_kwDOFIyF_M4C5alU",
        "name": "openwrt_h6_vplus_R21.11.11_k5.14.16-flippy-66+.7z",
        "label": null,
        "uploader": {
          "login": "breakings",
          "id": 25475074,
          "node_id": "MDQ6VXNlcjI1NDc1MDc0",
          "avatar_url": "https://avatars.githubusercontent.com/u/25475074?v=4",
          "gravatar_id": "",
          "url": "https://api.github.com/users/breakings",
          "html_url": "https://github.com/breakings",
          "followers_url": "https://api.github.com/users/breakings/followers",
          "following_url": "https://api.github.com/users/breakings/following{/other_user}",
          "gists_url": "https://api.github.com/users/breakings/gists{/gist_id}",
          "starred_url": "https://api.github.com/users/breakings/starred{/owner}{/repo}",
          "subscriptions_url": "https://api.github.com/users/breakings/subscriptions",
          "organizations_url": "https://api.github.com/users/breakings/orgs",
          "repos_url": "https://api.github.com/users/breakings/repos",
          "events_url": "https://api.github.com/users/breakings/events{/privacy}",
          "received_events_url": "https://api.github.com/users/breakings/received_events",
          "type": "User",
          "site_admin": false
        },
        "content_type": "application/x-7z-compressed",
        "state": "uploaded",
        "size": 338478639,
        "download_count": 12,
        "created_at": "2021-11-04T10:20:37Z",
        "updated_at": "2021-11-04T10:23:23Z",
        "browser_download_url": "https://github.com/breakings/OpenWrt/releases/download/flippy_openwrt/openwrt_h6_vplus_R21.11.11_k5.14.16-flippy-66%2B.7z"
      },
```

上面是一个节点的完整内容，改为只剩下面这几行即可
代码的其他部分，在保证文件格式正确的情况下均可删除，插件下载固件就用到了这几个内容。
在内核和插件的下载中也是一样的方式，用github的文件，简化一下即可用
```
    "tag_name": "flippy_openwrt",
    "assets": [
      {
        "created_at": "2021-11-04T10:20:37Z",
        "updated_at": "2021-11-04T10:23:23Z",
        "browser_download_url": "https://github.com/breakings/OpenWrt/releases/download/flippy_openwrt/openwrt_h6_vplus_R21.11.11_k5.14.16-flippy-66%2B.7z"
      },

```