ophub/luci-app-amlogic #14: 在线升级地址添加部分通配符支持？

Tao173, 2022-05-30:
很棒的项目，升级安装很方便，在使用中有一个需求：

使用action生成镜像到release，我编译了三种镜像，双L大和天灵的，我拒绝了生成多个tag和多个release，把它们合并到同一个release，

但是在宝盒里面升级就一直提示失败，release链接 [链接](https://github.com/Tao173/Auto-Update-Router/releases/tag/untagged-4fe9054e08e210242e60)
还请帮忙看看。

[image]

ophub, 2022-05-30:
仓库的releases地址发一下，插件里的设置截图发一下。/etc/config/amlogic 文件发一下，配置填写对了就可以使用。

Tao173, 2022-05-30:
好的，我发现问题了，为了release置顶，我开了Draft上传，搞的地址一直在变。谢谢回复。

closed 2022-05-30 by Tao173

Tao173, 2022-05-30:
> SOC

我是个zz，我重新发布了release，现在可以检测到更新了，但是我不知道它匹配的哪一个版本固件，因为三个版本的固件我都放在一个里面，详见release，还麻烦你帮忙看看。
[链接链接链接防失效](https://github.com/Tao173/Auto-Update-Router/releases/tag/ARMv8_Plus_Myself)

ophub, 2022-05-30:
下载最新的（最后上传到releases里面的，时间最新的）

Tao173, 2022-05-30:
ok，谢谢解答🤝