ophub/luci-app-amlogic #20: 安装出现 Not mounted successfully, try again。

iseeyou, 2022-09-05:
上午编译后在7月份版本成功升级
下午重新编译后继续安装就出现了错误。

FDT Value [ meson-gxl-s905d-phicomm-n1.dtb ]
K510 [ 0 ]
A total of [ 4 ] old partitions on EMMC will be deleted
Welcome to fdisk (util-linux 2.38).
Changes will remain in memory only, until you decide to write them.
Be careful before using the write command.

Command (m for help): Partition number (1-4, default 4): 
Partition 4 has been deleted.

Command (m for help): Partition number (1-3, default 3): 
Partition 3 has been deleted.

Command (m for help): Partition number (1,2, default 2): 
Partition 2 has been deleted.

Command (m for help): Selected partition 1
Partition 1 has been deleted.

Command (m for help): Partition type
   p   primary (0 primary, 0 extended, 4 free)
   e   extended (container for logical partitions)
Select (default p): Partition number (1-4, default 1): First sector (2048-15269887, default 2048): Last sector, +/-sectors or +/-size{K,M,G,T,P} (139264-15269887, default 15269887): 
Created a new partition 1 of type 'Linux' and of size 512 MiB.

Command (m for help): Partition type
   p   primary (1 primary, 0 extended, 3 free)
   e   extended (container for logical partitions)
Select (default p): Partition number (2-4, default 2): First sector (2048-15269887, default 2048): Last sector, +/-sectors or +/-size{K,M,G,T,P} (1638400-15269887, default 15269887): 
Created a new partition 2 of type 'Linux' and of size 960 MiB.

Command (m for help): Partition type
   p   primary (2 primary, 0 extended, 2 free)
   e   extended (container for logical partitions)
Select (default p): Partition number (3,4, default 3): First sector (2048-15269887, default 2048): Last sector, +/-sectors or +/-size{K,M,G,T,P} (3604480-15269887, default 15269887): 
Created a new partition 3 of type 'Linux' and of size 960 MiB.

Command (m for help): Partition type
   p   primary (3 primary, 0 extended, 1 free)
   e   extended (container for logical partitions)
Select (default e): 
Selected partition 4
First sector (2048-15269887, default 2048): Last sector, +/-sectors or +/-size{K,M,G,T,P} (5570560-15269887, default 15269887): 
Created a new partition 4 of type 'Linux' and of size 4.6 GiB.

Command (m for help): Partition number (1-4, default 4): Hex code or alias (type L to list all): 
Changed type of partition 'Linux' to 'W95 FAT32 (LBA)'.

Command (m for help): Partition number (1-4, default 4): Hex code or alias (type L to list all): 
Changed type of partition 'Linux' to 'Linux'.

Command (m for help): Partition number (1-4, default 4): Hex code or alias (type L to list all): 
Changed type of partition 'Linux' to 'Linux'.

Command (m for help): Partition number (1-4, default 4): Hex code or alias (type L to list all): 
Changed type of partition 'Linux' to 'Linux'.

Command (m for help): The partition table has been altered.
Calling ioctl() to re-read partition table.
Syncing disks.

Partition complete.
1+0 records in
1+0 records out
1+0 records in
1+0 records out
1+0 records in
1+0 records out
1+0 records in
1+0 records out
Write Android bootloader: [ /lib/u-boot/u-boot-2015-phicomm-n1.bin ]
444+0 records in
444+0 records out
8191+0 records in
8191+0 records out
Start creating file system ... 
Create a boot file system ... 
format boot partiton...
mkfs.fat 4.2 (2021-01-31)
Cannot initialize conversion from codepage 850 to UTF-8: No error information
Cannot initialize conversion from UTF-8 to codepage 850: No error information
Using internal CP850 conversion table
format rootfs1 partiton...
/usr/sbin/openwrt-install-amlogic: line 472: mkfs.btrfs: command not found
format rootfs2 partiton...
/usr/sbin/openwrt-install-amlogic: line 479: mkfs.btrfs: command not found
Wait for the boot file system to mount ... 
Successfully mounted.
copy boot ...
Edit uEnv.txt ...
complete.
complete.
Wait for the rootfs file system to mount ... 
Not mounted successfully, try again ...
Not mounted successfully, try again ...
Not mounted successfully, try again ...
Not mounted successfully, try again ...
Not mounted successfully, try again ...
Not mounted successfully, try again ...
Not mounted successfully, try again ...
Not mounted successfully, try again ...
Not mounted successfully, try again ...
Cannot mount rootfs file system, give up!

iseeyou, 2022-09-05:
刚才重新刷回了老版本，老版本可以升级到下午编译的。
今天编译的两个版本不能相互升级，只能刷回老版本再刷新版本

ophub, 2022-09-05:
> /usr/sbin/openwrt-install-amlogic: line 472: mkfs.btrfs: command not found

你缺这个

iseeyou, 2022-09-05:
> 

太感谢了。

closed 2022-09-05 by iseeyou