ophub/luci-app-amlogic #23: manual upload update: form token mismatch

leakssh, 2022-11-21:
`The submitted security token is invalid or already expired!`

In order to prevent unauthorized access to the system, your request has been blocked. Click "Continue »" below to return to the previous page.
[image]

ophub, 2022-11-21:
[image]

`Which repository's source code` is your openwrt compiled with? I use https://github.com/openwrt/openwrt and https://github.com/coolsnowwolf/lede test is fine.

leakssh, 2022-11-21:
i am using the repository from here https://github.com/immortalwrt/immortalwrt does it have any effect? 
I try to check all depends are available correctly

leakssh, 2022-11-21:
> [image]
> 
> `Which repository's source code` is your openwrt compiled with? I use https://github.com/openwrt/openwrt and https://github.com/coolsnowwolf/lede test is fine.

try to upload only the kernel not the full firmware, is it still normal?

ophub, 2022-11-21:
[image]

ophub, 2022-11-21:
/mnt/*4

leakssh, 2022-11-21:
> /mnt/*4
> 
> Is your openwrt system running on usb? Or is it written into emmc?

I run it on emmc, maybe something is wrong with this...
System Log:
`Mon Nov 21 22:03:29 2022 daemon.err uhttpd[5042]: Model amlogic/amlogic_info returns SimpleForm but is dispatched via cbi(),
Mon Nov 21 22:03:29 2022 daemon.err uhttpd[5042]: please change admin/system/amlogic to use the form() action instead.
Mon Nov 21 22:03:33 2022 daemon.err uhttpd[5042]: Model amlogic/amlogic_upload returns SimpleForm but is dispatched via cbi(),
Mon Nov 21 22:03:33 2022 daemon.err uhttpd[5042]: please change admin/system/amlogic/upload to use the form() action instead.
Mon Nov 21 22:03:33 2022 daemon.err uhttpd[5042]: Model amlogic/amlogic_upload returns SimpleForm but is dispatched via cbi(),
Mon Nov 21 22:03:33 2022 daemon.err uhttpd[5042]: please change admin/system/amlogic/upload to use the form() action instead.`

ophub, 2022-11-21:
It seems that immortalwrt has different syntax requirements. I have time to compile a system and debug it.

closed 2022-11-22 by ophub