ophub/luci-app-amlogic #24: The fdisk partition fails, Please try again

leakssh, 2022-11-25:
every time you install to emmc for the first time, a warning like this will appear:
`The kernel still uses the old table. The new table will be used at the next reboot or after you run partprobe(8) or partx(8).`

`The fdisk partition fails, Please try again.`

after trying the second time it just worked. what's wrong with this? and this always happens when you want to install to emmc you have to do it twice and try to succeed

ophub, 2022-11-25:
`cat /etc/flippy-openwrt-release`
What device, what soc, take a screenshot and post your error

leakssh, 2022-11-25:
`$ cat /etc/flippy-openwrt-release`
```
PLATFORM='amlogic'
FAMILY='meson-gxl'
BOARD='s905x'
KERNEL_VERSION='5.15.79'
K510='0'
FDTFILE='meson-gxl-s905x-p212.dtb'
UBOOT_OVERLOAD='u-boot-p212.bin'
MAINLINE_UBOOT=''
ANDROID_UBOOT=''
SOC='s905x'
```

ophub, 2022-11-25:
What box, error screenshot during operation

closed 2022-11-27 by ophub