ophub/luci-app-amlogic #25: 请问怎么自定义固件源地址呢？

cjlhll, 2022-12-09:
我照如下改了。没生效啊。files/etc/config/amlogic文件
```
config amlogic 'config'
	option enable '1'
	option amlogic_firmware_suffix '.img.gz'
	option amlogic_firmware_config '1'
	option amlogic_write_bootloader '1'
	option amlogic_firmware_repo 'https://github.com/cjlhll/N1-OpenWrt'
	option amlogic_kernel_branch '5.15'
	option amlogic_kernel_path 'https://github.com/breakings/OpenWrt/tree/main/opt/kernel'
	option amlogic_firmware_tag 'openwrt'
```

ophub, 2022-12-09:
https://github.com/ophub/luci-app-amlogic#custom-config

在你的diy2.sh里添加修改代码。

cjlhll, 2022-12-09:
> https://github.com/ophub/luci-app-amlogic#custom-config
> 
> 在你的diy2.sh里添加修改代码。

就不需要这个amlogic文件了是吗？只用在diy2.sh里添加这部分代码就行了。

ophub, 2022-12-09:
https://github.com/cjlhll/N1-OpenWrt/blob/master/.github/workflows/build-openwrt-N1.yml

```
    - name: Load custom configuration
      run: |
        [ -e files ] && mv files openwrt/files
        [ -e $CONFIG_FILE ] && mv $CONFIG_FILE openwrt/.config
        chmod +x $DIY_P2_SH
        cd openwrt
        $GITHUB_WORKSPACE/$DIY_P2_SH
```

你看你现在的流程，先把files拷贝进去了，但是最后一行被DIY_P2_SH脚本svn又覆盖成luci-app-amlogic的原版config了。
所以你拷贝的files无效。
所以根据你的流程代码，只有在DIY_P2_SH里处理才是有效的。

cjlhll, 2022-12-09:
> https://github.com/cjlhll/N1-OpenWrt/blob/master/.github/workflows/build-openwrt-N1.yml
>
> ```
>     - name: Load custom configuration
>       run: |
>         [ -e files ] && mv files openwrt/files [...]

明白了。那我把这个文件删掉，在diy2.sh里配置一下。
是这样配置吗？
[image]

ophub, 2022-12-09:
第一个网址对就可以，写你的仓库。
第二个是release里tags的关键字，只要保障你的tags里包含这个词就可以，而且其他x86等固件里不要包含这个词就可以。
第三个没变化可以不替换。

ophub, 2022-12-09:
对

cjlhll, 2022-12-09:
> 对

收到。非常感谢。~

closed 2022-12-09 by cjlhll

ophub, 2022-12-09:
比如有的人，一个仓库打包x86的时候他的tags里叫openwrt_x86-64_xxxx
打包N1也叫openwrt_xxx就不行了，这个openwrt关键词无法锁定是N1，这时要给N1起个和其他架构不重复的，比如叫
openwrt_s9xxx_xxx （关键词是s9xxx）  或者  openwrt_N1_xxx (关键词是N1)

cjlhll, 2022-12-09:
> 比如有的人，一个仓库打包x86的时候他的tags里叫openwrt_x86-64_xxxx 打包N1也叫openwrt_xxx就不行了，这个openwrt关键词无法锁定是N1，这时要给N1起个和其他架构不重复的，比如叫 openwrt_s9xxx_xxx （关键词是s9xxx） 或者 openwrt_N1_xxx (关键词是N1)

我明白了。得是唯一关键词。非常感谢作者大大的及时回答。

ophub, 2022-12-09:
不客气

cjlhll, 2022-12-12:
> 不客气

你好，地址是改过来了，但是为什么检测到的升级不对劲呢，不是我的仓库的更新，时间对不上，我尝试更新后就打不开了。整个op的后台管理页面都报错了。忘了截图了。

ophub, 2022-12-12:
https://github.com/ophub/luci-app-amlogic/blob/9cd50cf99e10f77b69808befae8d73dc9a6a3aae/luci-app-amlogic/root/usr/share/amlogic/amlogic_check_firmware.sh#L141

ophub, 2022-12-12:
curl -s "https://api.github.com/repos/你的用户/仓库地址/releases"

你看一下返回来的结果就知道了。
github的api使用美国时区（我没找到可以改时区的方法），所以返回的时候差8个时区。不是北京时间，但宝盒查询找出的肯定是最新的。结果正确。

目前支持使用这两个仓库的源码编译的tab包，其他仓库的源码可能不行，没适配过。
[OpenWrt](https://github.com/openwrt/openwrt)
[coolsnowwolf/lede](https://github.com/coolsnowwolf/lede)

cjlhll, 2022-12-13:
[image]
[image]
我怎么避免检查到此更新呢。

cjlhll, 2022-12-13:
> [image] [image] 我怎么避免检查到此更新呢。

确实是相差8小时

ophub, 2022-12-14:
上面的在工作流yml里可以设置时区，下面api的不知道怎么设置。
但是这不影响查询最新固件的结果。

cjlhll, 2022-12-14:
> 上面的在工作流yml里可以设置时区，下面api的不知道怎么设置。 但是这不影响查询最新固件的结果。

那就奇怪了。我特意把release删空了。然后重新编译了一次。刷入后还是能检测到更新。刚开始检测不到，过8小时就能检测到了。

ophub, 2022-12-14:
删tags

cjlhll, 2022-12-14:
> 删tags

删完了。release和tags都删干净了。测试的。