ophub/luci-app-amlogic #27: R68S 提示 Cannot find the partition corresponding to the root file system!

Jackxwb, 2022-12-25:
系统：openwrt-rockchip-R22.7.16-fastrhino-r68s-emmc-usbflash-img.rar
(地址：https://github.com/coolsnowwolf/lede/releases , 瑞芯微软件线刷的)

安装日志
```
root@OpenWrt:~# bash onekey-install.sh
01. Query server version information.
01.01 Latest version: 3.1.149-1
02. Check the latest plug-in download address.
02.01 Start downloading the latest plugin...
--2022-12-25 23:53:41--  https://github.com/ophub/luci-app-amlogic/releases/download/3.1.149-1/luci-app-amlogic_3.1.149-1_all.ipk
Resolving github.com... 20.205.243.166
Connecting to github.com|20.205.243.166|:443... connected.
HTTP request sent, awaiting response... No data received.
Retrying.

--2022-12-25 23:55:42--  (try: 2)  https://github.com/ophub/luci-app-amlogic/releases/download/3.1.149-1/luci-app-amlogic_3.1.149-1_all.ipk
Connecting to github.com|20.205.243.166|:443... connected.
HTTP request sent, awaiting response... 302 Found
Location: https://objects.githubusercontent.com/github-production-release-asset-2e65be/376685891/c682237a-bddb-4b3d-8b1f-e07e23f31538?X-Amz-Algorithm=AWS4-HMAC-SHA256&X-Amz-Credential=AKIAIWNJYAX4CSVEH53A%2F20221225%2Fus-east-1%2Fs3%2Faws4_request&X-Amz-Date=20221225T155558Z&X-Amz-Expires=300&X-Amz-Signature=a373030df44ad5fc2a0bdca805d82859534c23f6d99eb93a508b365640a898ed&X-Amz-SignedHeaders=host&actor_id=0&key_id=0&repo_id=376685891&response-content-disposition=attachment%3B%20filename%3Dluci-app-amlogic_3.1.149-1_all.ipk&response-content-type=application%2Foctet-stream [following]
--2022-12-25 23:55:58--  https://objects.githubusercontent.com/github-production-release-asset-2e65be/376685891/c682237a-bddb-4b3d-8b1f-e07e23f31538?X-Amz-Algorithm=AWS4-HMAC-SHA256&X-Amz-Credential=AKIAIWNJYAX4CSVEH53A%2F20221225%2Fus-east-1%2Fs3%2Faws4_request&X-Amz-Date=20221225T155558Z&X-Amz-Expires=300&X-Amz-Signature=a373030df44ad5fc2a0bdca805d82859534c23f6d99eb93a508b365640a898ed&X-Amz-SignedHeaders=host&actor_id=0&key_id=0&repo_id=376685891&response-content-disposition=attachment%3B%20filename%3Dluci-app-amlogic_3.1.149-1_all.ipk&response-content-type=application%2Foctet-stream
Resolving objects.githubusercontent.com... 185.199.109.133, 185.199.108.133, 185.199.111.133, ...
Connecting to objects.githubusercontent.com|185.199.109.133|:443... connected.
HTTP request sent, awaiting response... 200 OK
Length: 76939 (75K) [application/octet-stream]
Saving to: '/root/luci-app-amlogic_3.1.149-1_all.ipk'

/root/luci-app-amlogic_3.1.149-1_all.ipk                    100%[===========================================================================================================================================>]  75.14K   168KB/s    in 0.4s

2022-12-25 23:56:00 (168 KB/s) - '/root/luci-app-amlogic_3.1.149-1_all.ipk' saved [76939/76939]

02.02 luci-app-amlogic_3.1.149-1_all.ipk complete.
--2022-12-25 23:56:00--  https://github.com/ophub/luci-app-amlogic/releases/download/3.1.149-1/luci-i18n-amlogic-zh-cn_3.1.149-1_all.ipk
Resolving github.com... 20.205.243.166
Connecting to github.com|20.205.243.166|:443... connected.
HTTP request sent, awaiting response... 302 Found
Location: https://objects.githubusercontent.com/github-production-release-asset-2e65be/376685891/b6ebc705-0c79-4cd6-b4c6-8b50974f5626?X-Amz-Algorithm=AWS4-HMAC-SHA256&X-Amz-Credential=AKIAIWNJYAX4CSVEH53A%2F20221225%2Fus-east-1%2Fs3%2Faws4_request&X-Amz-Date=20221225T155705Z&X-Amz-Expires=300&X-Amz-Signature=69fb9ab32d5c89145b0bfe13ebbc55526c2ba91a3e00d12caf7df6608cef86bf&X-Amz-SignedHeaders=host&actor_id=0&key_id=0&repo_id=376685891&response-content-disposition=attachment%3B%20filename%3Dluci-i18n-amlogic-zh-cn_3.1.149-1_all.ipk&response-content-type=application%2Foctet-stream [following]
--2022-12-25 23:57:05--  https://objects.githubusercontent.com/github-production-release-asset-2e65be/376685891/b6ebc705-0c79-4cd6-b4c6-8b50974f5626?X-Amz-Algorithm=AWS4-HMAC-SHA256&X-Amz-Credential=AKIAIWNJYAX4CSVEH53A%2F20221225%2Fus-east-1%2Fs3%2Faws4_request&X-Amz-Date=20221225T155705Z&X-Amz-Expires=300&X-Amz-Signature=69fb9ab32d5c89145b0bfe13ebbc55526c2ba91a3e00d12caf7df6608cef86bf&X-Amz-SignedHeaders=host&actor_id=0&key_id=0&repo_id=376685891&response-content-disposition=attachment%3B%20filename%3Dluci-i18n-amlogic-zh-cn_3.1.149-1_all.ipk&response-content-type=application%2Foctet-stream
Resolving objects.githubusercontent.com... 185.199.110.133, 185.199.111.133, 185.199.108.133, ...
Connecting to objects.githubusercontent.com|185.199.110.133|:443... connected.
HTTP request sent, awaiting response... 200 OK
Length: 6564 (6.4K) [application/octet-stream]
Saving to: '/root/luci-i18n-amlogic-zh-cn_3.1.149-1_all.ipk'

/root/luci-i18n-amlogic-zh-cn_3.1.149-1_all.ipk             100%[===========================================================================================================================================>]   6.41K  --.-KB/s    in 0.001s

2022-12-25 23:57:06 (11.5 MB/s) - '/root/luci-i18n-amlogic-zh-cn_3.1.149-1_all.ipk' saved [6564/6564]

02.03 luci-i18n-amlogic-zh-cn_3.1.149-1_all.ipk complete.
03. Start automatic installation.
No packages removed.
Installing luci-app-amlogic (3.1.149-1) to root...
Installing dosfstools (4.2-1) to root...
Downloading https://downloads.immortalwrt.org/releases/packages-18.06-k5.4/aarch64_generic/packages/dosfstools_4.2-1_aarch64_generic.ipk
Installing pv (1.6.6-1) to root...
Downloading https://downloads.immortalwrt.org/releases/packages-18.06-k5.4/aarch64_generic/packages/pv_1.6.6-1_aarch64_generic.ipk
Installing losetup (2.38-12) to root...
Downloading https://downloads.immortalwrt.org/releases/packages-18.06-k5.4/aarch64_generic/base/losetup_2.38-12_aarch64_generic.ipk
Installing uuidgen (2.38-12) to root...
Downloading https://downloads.immortalwrt.org/releases/packages-18.06-k5.4/aarch64_generic/base/uuidgen_2.38-12_aarch64_generic.ipk
Installing perl (5.28.1-7) to root...
Downloading https://downloads.immortalwrt.org/releases/packages-18.06-k5.4/aarch64_generic/packages/perl_5.28.1-7_aarch64_generic.ipk
Installing libfdisk1 (2.38-12) to root...
Downloading https://downloads.immortalwrt.org/releases/packages-18.06-k5.4/aarch64_generic/base/libfdisk1_2.38-12_aarch64_generic.ipk
Installing fdisk (2.38-12) to root...
Downloading https://downloads.immortalwrt.org/releases/packages-18.06-k5.4/aarch64_generic/base/fdisk_2.38-12_aarch64_generic.ipk
Installing luci-i18n-amlogic-zh-cn (3.1.149-1) to root...
Configuring dosfstools.
Configuring pv.
Configuring uuidgen.
Configuring losetup.
Configuring perl.
Configuring libfdisk1.
Configuring fdisk.
Configuring luci-app-amlogic.
Can't locate strict.pm in @INC (you may need to install the strict module) (@INC contains: /usr/lib/perl5/5.28) at /usr/sbin/fixcpufreq.pl line 3.
BEGIN failed--compilation aborted at /usr/sbin/fixcpufreq.pl line 3.
Configuring luci-i18n-amlogic-zh-cn.
03.01 The plugin has been installed successfully, Path: System -> Amlogic Service.
```

在“在线下载更新”里面，三个更新按钮点击后都提示“Cannot find the partition corresponding to the root file system!”
[image]

在手动上传更新里面，上传文件升级会出现“/etc/flippy-openwrt-release file is missing!”，参数Issues往期记录，我执行了`echo "PLATFORM='amlogic'" >/etc/flippy-openwrt-release`这句，点击升级后会闪过一句“Current device: FastRhino R68S [ amlogic ]”，然后错误信息也变成了“Cannot find the partition corresponding to the root file system!”
[image]

Jackxwb, 2022-12-25:
[image]
挂载点截图
[image]
硬盘截图
[image]

ophub, 2022-12-26:
晶晨宝盒插件仅适用于f大的openwrt系统，不适合其他人发布的系统。
因为f大的openwrt分区、内核等都和他们不一样，所以更新固件、更新内核这些功能在其他系统都无法使用。

closed 2022-12-26 by ophub