ophub/luci-app-amlogic #29: N1怎么装这个插件呢 用一键脚本报错

nhtcool, 2023-01-02:
Collected errors:
 * pkg_hash_check_unresolved: cannot find dependency pv for luci-app-amlogic
 * pkg_hash_fetch_best_installation_candidate: Packages for luci-app-amlogic found, but incompatible with the architectures configured
 * opkg_install_cmd: Cannot install package luci-app-amlogic.
 * satisfy_dependencies_for: Cannot satisfy the following dependencies for luci-i18n-amlogic-zh-cn:
 *      pv
 * opkg_install_cmd: Cannot install package luci-i18n-amlogic-zh-cn.
03.01 The plugin has been installed successfully, Path: System -> Amlogic Service.

ophub, 2023-01-02:
https://github.com/ophub/amlogic-s9xxx-openwrt/blob/main/router-config/README.cn.md#1011-openwrt-%E5%BF%85%E9%80%89%E9%A1%B9

10.11里面有必选依赖对照表，你的错误提示没有pv这个软件包。你可以在系统，软件管理里，先点更新软件列表，再搜索pv，点安装，等安装好pv后再用一键脚本安装。

nhtcool, 2023-01-02:
谢谢 就是固件编译不是很懂，现在这个固件觉得好用，就是内核比较老了 想用您这个工具更新内核，按照您说的安装了PV后显示了以下错误，是不是要在5.15.85这个内核上才能安装。
pkg_hash_check_unresolved: cannot find dependency kernel (= 5.15.85-1-228b3409cc0192f42da9fd813f2c3880) for kmod-nfnetlink
 * pkg_hash_check_unresolved: cannot find dependency kernel (= 5.15.85-1-228b3409cc0192f42da9fd813f2c3880) for kmod-nf-reject
 * pkg_hash_check_unresolved: cannot find dependency kernel (= 5.15.85-1-228b3409cc0192f42da9fd813f2c3880) for kmod-nf-reject6
 * pkg_hash_check_unresolved: cannot find dependency kernel (= 5.15.85-1-228b3409cc0192f42da9fd813f2c3880) for kmod-nf-conntrack
 * pkg_hash_check_unresolved: cannot find dependency kernel (= 5.15.85-1-228b3409cc0192f42da9fd813f2c3880) for kmod-nf-conntrack6
 * pkg_hash_check_unresolved: cannot find dependency kernel (= 5.15.85-1-228b3409cc0192f42da9fd813f2c3880) for kmod-nf-nat
 * pkg_hash_check_unresolved: cannot find dependency kernel (= 5.15.85-1-228b3409cc0192f42da9fd813f2c3880) for kmod-nf-log
 * pkg_hash_fetch_best_installation_candidate: Packages for kmod-nf-log found, but incompatible with the architectures configured
 * pkg_hash_check_unresolved: cannot find dependency kernel (= 5.15.85-1-228b3409cc0192f42da9fd813f2c3880) for kmod-nf-log6
 * pkg_hash_fetch_best_installation_candidate: Packages for kmod-nf-log6 found, but incompatible with the architectures configured
 * pkg_hash_check_unresolved: cannot find dependency kernel (= 5.15.85-1-228b3409cc0192f42da9fd813f2c3880) for kmod-crypto-hash
 * pkg_hash_check_unresolved: cannot find dependency kernel (= 5.15.85-1-228b3409cc0192f42da9fd813f2c3880) for kmod-crypto-crc32c
 * pkg_hash_check_unresolved: cannot find dependency kernel (= 5.15.85-1-228b3409cc0192f42da9fd813f2c3880) for kmod-lib-crc32c
 * pkg_hash_check_unresolved: cannot find dependency kernel (= 5.15.85-1-228b3409cc0192f42da9fd813f2c3880) for kmod-nft-core
 * pkg_hash_fetch_best_installation_candidate: Packages for kmod-nft-core found, but incompatible with the architectures configured
 * pkg_hash_check_unresolved: cannot find dependency kernel (= 5.15.85-1-228b3409cc0192f42da9fd813f2c3880) for kmod-nft-fib
 * pkg_hash_fetch_best_installation_candidate: Packages for kmod-nft-fib found, but incompatible with the architectures configured
 * pkg_hash_check_unresolved: cannot find dependency kernel (= 5.15.85-1-228b3409cc0192f42da9fd813f2c3880) for kmod-nf-flow
 * pkg_hash_check_unresolved: cannot find dependency kernel (= 5.15.85-1-228b3409cc0192f42da9fd813f2c3880) for kmod-nft-nat
 * pkg_hash_fetch_best_installation_candidate: Packages for kmod-nft-nat found, but incompatible with the architectures configured
 * pkg_hash_check_unresolved: cannot find dependency kernel (= 5.15.85-1-228b3409cc0192f42da9fd813f2c3880) for kmod-nft-offload
 * pkg_hash_fetch_best_installation_candidate: Packages for kmod-nft-offload found, but incompatible with the architectures configured
 * pkg_hash_check_unresolved: cannot find dependency kernel (= 5.15.85-1-228b3409cc0192f42da9fd813f2c3880) for kmod-nf-ipt
 * pkg_hash_check_unresolved: cannot find dependency kernel (= 5.15.85-1-228b3409cc0192f42da9fd813f2c3880) for kmod-ipt-core
 * pkg_hash_check_unresolved: cannot find dependency kernel (= 5.15.85-1-228b3409cc0192f42da9fd813f2c3880) for kmod-ipt-conntrack
 * pkg_hash_check_unresolved: cannot find dependency kernel (= 5.15.85-1-228b3409cc0192f42da9fd813f2c3880) for kmod-ipt-nat
03.01 The plugin has been installed successfully, Path: System -> Amlogic Service.

ophub, 2023-01-02:
和内核无关，你现在到你的系统菜单下，刷新看看是不是已经安装好了晶晨宝盒。

pv这个软件包是使用ddbr时进度条的依赖项，如果你不使用这个功能没影响。

nhtcool, 2023-01-02:
果然，有了，非常感谢。

closed 2023-01-02 by ophub