ophub/luci-app-amlogic #31: 关于宝盒快照

nhtcool, 2023-02-14:
首先谢谢大佬，又来打搅你了，请问晶晨宝盒里面的快照备份和系统自带的备份有什么区别呢，如果用快照还原的话，必须也要相同固件和相同插件的情况下才可以用吗？

ophub, 2023-02-14:
op系统自带的备份与还原功能应该是都可以使用，但是升级功能不支持盒子使用。

晶晨宝盒插件是为f大的固件量身定制的，和op里面的其他的各种插件一样，这个插件本身只是luci可视化的操作界面，全部的操作执行都是调用各脚本来执行的，包括安装到emmc,更新内核，更新完整固件，创建快照，还还快照，备份与恢复，CPU设置等等全部功能。插件只是把敲命令变成了点按钮，所以在插件的可视化界面操作的结果和在终端里敲命令执行的过程和结果都是完全相同的。

备份/恢复一般建议用在版本跨度不太大的固件之间，如75的备份在80里还原是可以的，但是50的备份在80里还原，可能个别的插件/etc/config发生了根本性变化，已经无法在新版里使用了。

快照和系统相关，只能在当前系统使用，安装/更新完整op系统时会自动生成这两个节点的快照。自己平时可以点创建快照生成更多个记录点。（如果你玩游戏的话，单机游戏里有存档功能，就是把当前的进度保存下来，以后可以从此刻开始继续游戏。）快照就是这个作用，把当前的系统生成一个记录。然后你继续设置、测试、重启调试等，如果又搞好一些插件，就再创建一个快照。当你什么时候搞砸了，op不能正常工作时，你就到快照里，选择一个你曾经认为很好的快照点，恢复回去。这就是快照的意思，和在帝国时代游戏里存档了多个进度一样。它不能像备份的配置那样下载用于其他系统，只能在当前系统里使用。快照只会记录差异，存档记录很小，可以多点多用，多给执腾埋几个后悔药，随时一键回到解放前。

nhtcool, 2023-02-15:
谢谢大佬如此完整的说明 让我等小白能轻易看懂 快照功能如此方便，多记录点总是好事，为了系统更加完整备份，也参考了您提供的ddbr备份方法，备份了完整系统，也就不怕系统抽风了

closed 2023-02-16 by ophub