ophub/luci-app-amlogic #35: 更新全系统失败，提示：Resource busy

expliyh, 2023-04-08:
完整日志：
Sat Apr  8 16:12:56 2023 daemon.err uhttpd[4099]: sh: can't create /tmp/.luci-app-amlogic/op_release_code: nonexistent directory
Sat Apr  8 16:12:59 2023 kern.info kernel: [  193.409755] loop0: detected capacity change from 0 to 2328576
Sat Apr  8 16:12:59 2023 kern.info kernel: [  193.411066]  loop0: p1 p2
Sat Apr  8 16:13:02 2023 kern.info kernel: [  196.470395] EXT4-fs (loop0p1): mounted filesystem with ordered data mode. Quota mode: none.
Sat Apr  8 16:13:02 2023 kern.info kernel: [  196.473976] BTRFS: device label EMMC_ROOTFS1 devid 1 transid 16 /dev/loop0p2 scanned by mount (29859)
Sat Apr  8 16:13:02 2023 kern.info kernel: [  196.475441] BTRFS info (device loop0p2): using crc32c (crc32c-generic) checksum algorithm
Sat Apr  8 16:13:02 2023 kern.info kernel: [  196.476202] BTRFS info (device loop0p2): use zstd compression, level 6
Sat Apr  8 16:13:02 2023 kern.info kernel: [  196.476783] BTRFS info (device loop0p2): using free space tree
Sat Apr  8 16:13:02 2023 kern.info kernel: [  196.481873] BTRFS info (device loop0p2): enabling ssd optimizations
Sat Apr  8 16:13:02 2023 daemon.err uhttpd[4099]: ERROR: unable to open /dev/mmcblk0p3: Resource busy
Sat Apr  8 16:13:02 2023 kern.info kernel: [  196.547327] EXT4-fs (loop0p1): unmounting filesystem.

ophub, 2023-04-08:
系统占用中，重启再试

expliyh, 2023-04-08:
> 系统占用中，重启再试

这是重启后的，并且报错以后用lsof检查，发现没有占用这个设备的进程

另外用 https://github.com/haiibo/OpenWrt 的系统在 r68s上测试出现如下错误，系统为全新安装。
Fri Apr  7 23:20:31 2023 daemon.err uhttpd[4035]: sh: can't create /tmp/.luci-app-amlogic/op_release_code: nonexistent directoryFri Apr  7 23:20:59 2023 kern.info kernel: [  266.663763] loop0: detected capacity change from 0 to 2328576Fri Apr  7 23:20:59 2023 kern.info kernel: [  266.665068]  loop0: p1 p2Fri Apr  7 23:21:02 2023 kern.info kernel: [  269.731493] EXT4-fs (loop0p1): mounted filesystem with ordered data mode. Quota mode: none.Fri Apr  7 23:21:02 2023 daemon.err uhttpd[4035]: mount: mounting /dev/loop0p2 on /mnt/mmcblk0p4/root failed: File existsFri Apr  7 23:21:02 2023 kern.warn kernel: [  269.735748] BTRFS warning: duplicate device /dev/loop0p2 devid 1 generation 15 scanned by mount (27100)Fri Apr  7 23:21:02 2023 kern.info kernel: [  269.741310] EXT4-fs (loop0p1): unmounting filesystem.

ophub, 2023-04-08:
你的4个分区是什么？
`lsblk`

expliyh, 2023-04-10:
> 你的4个分区是什么？ `lsblk`

[image]

closed 2023-04-17 by ophub