ophub/luci-app-amlogic #36: 当在原Release记录上更新时，无法检查更新

Gzxhwq, 2023-04-10:
https://github.com/ophub/luci-app-amlogic/blob/f13123035bf59d714541410f34347bec6ceffbe2/luci-app-amlogic/root/usr/share/amlogic/amlogic_check_firmware.sh#L143

例如：https://github.com/Gzxhwq/OpenWrt/releases

把 .published_at改成.assets[].updated_at 是否可以？

ophub, 2023-04-10:
我看了几遍才理解了你的意思，你是说tags名字永远不变，所以这个.published_at就不会变了，所以脚本对比了日期，检查没有新文件是吗？

你看一下api返回的json文件，有你的这个节点吧？和生成的下载文件同时生成的匹配节点。我在外面调试不了，你把这个json文件此处截图发一下。

这个文件可以在本地openwrt的/usr/share/amlogic/里面修改这个文件测试，可以的话你提交一个pr

Gzxhwq, 2023-04-10:
我的这个源的情况下，把 .published_at改成.assets[].updated_at 是可以的，但是不知道会不会影响其他部分，因为我不知道当初选检测published_at是否出于某种考虑。

而且以前检测是没问题的，我发现问题好像是最近的事。因为默认的固件源https://github.com/breakings/OpenWrt/releases 也是用这种方法更新release固件，应该也会有同样的情况。

ophub, 2023-04-10:
.published_at是tags的发布日期，多个包含相同关键词的tags用日期排序选择最新发布的

ophub, 2023-04-10:
插件更新了，取的固件的更新日期，不是tags的日期了，你把插件更新到最新版再试试

Gzxhwq, 2023-04-10:
好的，谢谢，应该可以了

closed 2023-04-10 by Gzxhwq