ophub/luci-app-amlogic #41: DG-3399晶晨宝盒显示设备型号为当前设备：PLATFORM: rockchip

LemonCrab666, 2023-07-17:
DG-3399晶晨宝盒显示设备型号为当前设备：PLATFORM: rockchip，在线检查晶晨宝盒更新显示The search results for releases are empty.

ophub, 2023-07-17:
插件设置里填写的什么，截图发下

LemonCrab666, 2023-07-17:
[image]

LemonCrab666, 2023-07-17:
[image]

LemonCrab666, 2023-07-17:
[image]

ophub, 2023-07-17:
在线下载更新，先点第一个按钮（升级插件），等执行完了，再点第三个按钮（更新完整固件）

LemonCrab666, 2023-07-17:
> 在线下载更新，先点第一个按钮（升级插件），等执行完了，再点第三个按钮（更新完整固件）

我是自编译的，用你的打包出来的

ophub, 2023-07-19:
你把设置里的固件地址改成你的仓库

closed 2023-07-27 by ophub

LemonCrab666, 2023-09-17:
直接选择本地img.gz文件提示Unknown device: [ DG-3399 ], Not supported.dtb用的也是仓库里提供的

LemonCrab666, 2023-09-17:
> 你把设置里的固件地址改成你的仓库
直接选择本地img.gz文件提示Unknown device: [ DG-3399 ], Not supported.dtb用的也是仓库里提供的

LemonCrab666, 2023-09-17:
> 你把设置里的固件地址改成你的仓库

[image]