ophub/luci-app-amlogic #44: 无法支持新版LuCI的上传功能

OldCoding, 2023-08-30:
在immortalwrt中无法正常上传，使用的是master分支，回复的是“新 LuCI 已不支持传统 lua 插件的文件上传功能，重写为 js 即可。”，望更新一下以支持新版LuCI
[image]

ophub, 2023-09-02:
https://github.com/ophub/amlogic-s9xxx-openwrt/releases/tag/OpenWrt_immortalwrt_save_2023.08

我下载测试了可以上传，你使用的什么版本。去看了下他luci仓库的文件上传是1年前更新的，还是lua，你能找个JavaScript上传的app案例发我个地址我学习学习。

https://github.com/ophub/amlogic-s9xxx-openwrt/releases/tag/OpenWrt_immortalwrt_save_2023.08

[image]

[image]

[image]

OldCoding, 2023-09-02:
> https://github.com/ophub/amlogic-s9xxx-openwrt/releases/tag/OpenWrt_immortalwrt_save_2023.08
> 
> 我下载测试了可以上传，你使用的什么版本。去看了下他luci仓库的文件上传是1年前更新的，还是lua，你能找个JavaScript上传的app案例发我个地址我学习学习。
> 
> https://github.com/immortalwrt/luci/tree/master/applications/luci-app-filetransfer
> 
> [image]
> 
> [image]
> 
> [image]

我用的是自编译的固件，https://github.com/OldCoding/openwrt_packit_arm/releases/tag/ARMv8-im，案例我这就没有了，天灵说在软件包的页面可以直接上传APK，你可以看下是否能找到相关的功能源码

Saxon-Sun, 2023-09-20:
@ophub 问下luci-app-amlogic支持openwrt luci 23.05分支的吗？

ophub, 2023-09-20:
支持

Saxon-Sun, 2023-09-21:
> 支持

谢谢大佬

closed 2023-09-28 by ophub