ophub/luci-app-amlogic #45: DG-3399晶晨宝盒显示设备型号为当前设备：PLATFORM: rockchip

LemonCrab666, 2023-09-17:
直接选择本地img.gz文件提示Unknown device: [ DG-3399 ], Not supported.dtb用的也是仓库里提供的
[image]
文件大小也不对

ophub, 2023-09-17:
当前设备：rockchip 是对的，这个地方会显示设备的平台，支持的设备会显示为 amlogic/allwinner/rockchip 三个平台。
文件在线上传不如ftp上传好。你使用ssh访问，上传文件到/mnt/*p4/ 的目录下，然后在此可以看到p4里上传的文件，然后点击更新即可。

LemonCrab666, 2023-09-17:
> 当前设备：rockchip 是对的，这个地方会显示设备的平台，支持的设备会显示为 amlogic/allwinner/rockchip 三个平台。 文件在线上传不如ftp上传好。你使用ssh访问，上传文件到/mnt/*p4/ 的目录下，然后在此可以看到p4里上传的文件，然后点击更新即可。

我刚看了下[openwrt-update-rockchip](https://github.com/ophub/luci-app-amlogic/blob/main/luci-app-amlogic/root/usr/sbin/openwrt-update-rockchip)这个文件，里面是没有rk3399的分类和机型的，是因为这个原因没办法正确识别机型吗

ophub, 2023-09-17:
那个地方识别的是平台PLATFORM，不是soc

LemonCrab666, 2023-09-17:
> 使用lede的源码编译的固件，上传基本上没问题。但是immortalwrt现在使用的JavaScript替代了lua，上传我发现不太好用了，需要改这个上传功能为JavaScript，我在纠结是以lede为主还是换。

我刚直接winscp上传了img文件，显示依旧是Unknown device: [ dg3399 ], Not supported.
[image]

ophub, 2023-09-17:
https://github.com/ophub/luci-app-amlogic/blob/547661e2dc3e50ea44b747057f5a70897bae7959/luci-app-amlogic/luasrc/model/cbi/amlogic/amlogic_config.lua#L24-L29

显示rockchip是对的。

LemonCrab666, 2023-09-17:
> https://github.com/ophub/luci-app-amlogic/blob/547661e2dc3e50ea44b747057f5a70897bae7959/luci-app-amlogic/luasrc/model/cbi/amlogic/amlogic_config.lua#L24-L29
> 
> 显示rockchip是对的。

机型显示也无所谓了，现在就是手动更新一直不成功，打包用的amlogic-s9xxx-openwrt

LemonCrab666, 2023-09-17:
> https://github.com/ophub/luci-app-amlogic/blob/547661e2dc3e50ea44b747057f5a70897bae7959/luci-app-amlogic/luasrc/model/cbi/amlogic/amlogic_config.lua#L24-L29
> 
> 显示rockchip是对的。

按照您说的办法也没解决，还有什么脚本可以直接更新吗

ophub, 2023-09-17:
更新固件的脚本要升级一下，支持从记录文件里读取更多自定义的盒子。明天我抽空升级一下。

LemonCrab666, 2023-09-17:
> 更新固件的脚本要升级一下，支持从记录文件里读取更多自定义的盒子。明天我抽空升级一下。

好的，明天我再试下

ophub, 2023-09-18:
升级插件到198试试

LemonCrab666, 2023-09-18:
> 升级插件到198试试

更新完，开不开机了……

ophub, 2023-09-18:
我测试正常。拔插电源看看。提供有用的信息。

LemonCrab666, 2023-09-18:
> 我测试正常。拔插电源看看。提供有用的信息。

重新插拔电源过了，设备灯正常亮，op没有启动，hdmi也没输出，刷写的时候勾选了bootloader，跟这个有关吗

ophub, 2023-09-18:
那个选项只对amlogic有用

LemonCrab666, 2023-09-19:
> 那个选项只对amlogic有用

我再不勾选试试看，可能是勾选这个选项导致引导开机吧，现在插上电源什么反应都没有，hdmi也没有输出

ophub, 2023-09-19:
勾不勾对你无差别，只对amlogic有效，和勾无关。

LemonCrab666, 2023-09-20:
> 勾不勾对你无差别，只对amlogic有效，和勾无关。

重新刷入之后，再次用198更新，出现图中提示
[image]

LemonCrab666, 2023-09-23:
> 勾不勾对你无差别，只对amlogic有效，和勾无关。

大佬，这个提示要怎么解决呢
[image]

ophub, 2023-09-23:
更新最新版插件，重启设备试试。

LemonCrab666, 2023-09-23:
> 更新最新版插件，重启设备试试。

更新了最新的插件201，也重启了设备，更新依然提示这个
[image]

LemonCrab666, 2023-09-23:
> 更新最新版插件，重启设备试试。

重新打包最新的内核，更新成功，但是重启设备hdmi无画面，可能无法引导系统

ophub, 2023-09-23:
去路由器里看看armbian设备有没有上线，看看ip

LemonCrab666, 2023-09-23:
> 去路由器里看看armbian设备有没有上线，看看ip

luci界面无法进入，也ping不到设备

LemonCrab666, 2023-09-23:
> 去路由器里看看armbian设备有没有上线，看看ip

重刷了一遍，依然是这个问题，更新后无法开机，无HDMI显示无法ping到设备，路由器也看不到设备上线

LemonCrab666, 2023-09-23:
仅更新内核也不成功
[image]
[image]
版本分支设置的没问题

ophub, 2023-09-23:
这个是无法查询服务器，先把op上网设置好了，让盒子可以正常访问github.com服务器。

LemonCrab666, 2023-09-24:
> 这个是无法查询服务器，先把op上网设置好了，让盒子可以正常访问github.com服务器。

网络设置没有问题，github服务器可以正常访问

closed 2023-09-28 by ophub