ophub/luci-app-amlogic #51: 求助安装到EMMc时自定义分区大小

nantayo, 2024-02-18:
我使用的设备是斐讯N1，原版的8G EMMc。在观察了分区容量足够的情况下，我试图减小boot和两个rootfs分区，以便给share分区留出大一点的空间。我尝试SCP登入后编辑/usr/sbin路径下的openwrt-install-amlogic文件，将#265至#272之间的内容更改为：

ROOT1="256"
ROOT2="256"
if [[ "${AMLOGIC_SOC}" == "s912" || "${AMLOGIC_SOC}" == "s905d" ]]; then
    BOOT="128"
    BLANK1="64"
    BLANK2="16"
    BLANK3="16"
    BLANK4="64"

但写入后boot和rootfs分区大小并未发生变化，仍然是默认的512M和960M。请指教写入EMMc时是否能够修改各个分区的大小，除了796-800M这一段安全位置以外是否还有某些特别的限制？如果能修改的话应该怎样操作？

ophub, 2024-02-22:
目前的分区是安全的，调整偏移位置可能会写入不安全区域变砖。在原版安卓系统下可以查看分区情况，在我的armbian仓库里有文档。
修改的方法就是你这样，计算起始和结束位置，修改这个文件。

nantayo, 2024-02-22:
> 目前的分区是安全的，调整偏移位置可能会写入不安全区域变砖。在原版安卓系统下可以查看分区情况，在我的armbian仓库里有文档。 修改的方法就是你这样，计算起始和结束位置，修改这个文件。

原来如此，看来需要学习的内容还有很多。
不过想请教下，luci界面的安装也是调用/usr/sbin/openwrt-install-amlogic吗？我修改这个文件后用luci界面的安装功能，修改值并没生效，不太清楚原因

ophub, 2024-02-22:
luci只是操作界面，调用的全部是.sh，包括安装/更新固件/更新内核/备份恢复等等全部都是调用的相同的脚本，界面操作的结果和在终端里使用命令的结果完全相同。

closed 2024-02-26 by nantayo