ophub/luci-app-amlogic #54: 使用 LuCI 界面更新固件及内核无法找到依赖，使用 openwrt-kernel 终端命令则正常

etnperlong, 2024-06-10:
作者大大你好👋

我在尝试使用 LuCI 界面更新内核时，会出现以下错误 
` [Error] Missing [ tar ] in OpenWrt firmware, unable to update kernel`

[image]

但如果我在失败之后，通过SSH终端输入 `openwrt-kernel` 指令，则可成功更新内核

经过测试，我发现无论是更新系统还是更新内核，均会出现 找不到依赖 的错误

检查代码后发现相关行在 https://github.com/ophub/luci-app-amlogic/blob/bcaf9d5b4f7b55adbf5bf75dcefab028e06c31c3/luci-app-amlogic/root/usr/sbin/openwrt-update-amlogic#L122
和 https://github.com/ophub/luci-app-amlogic/blob/bcaf9d5b4f7b55adbf5bf75dcefab028e06c31c3/luci-app-amlogic/root/usr/sbin/openwrt-kernel#L59

我的设备是 N1 盒子，运行的系统是 ImmortalWRT 23.05.2， 使用 [amlogic-s9xxx-openwrt](https://github.com/ophub/amlogic-s9xxx-openwrt) 中 ImageBuilder 方式构建，加入自定义的软件包

系统信息如下：
[Powered by LuCI openwrt-23.05 branch (git-24.086.26304-1d8fc03)](https://github.com/openwrt/luci)
[ArgonTheme v2.3.1](https://github.com/jerrykuku/luci-theme-argon)
ImmortalWrt 23.05.2 r27625-416c8c5c91

晶晨宝盒版本： 3.1.234-1

不清楚是哪个环节导致 LuCI 界面的指令失效，希望作者如果有时间的话，帮我除错一下，谢谢🙏
如果需要更多的详细信息，我可以提供

ophub, 2024-06-11:
把你使用ImageBuilder构建的openwrt下载地址发一下

etnperlong, 2024-06-11:
> 把你使用ImageBuilder构建的openwrt下载地址发一下

好的，我是通过本地构建的，镜像已经通过 OneDrive 临时分享
[链接](https://makeppt-my.sharepoint.com/:u:/g/personal/pan_makeppt_onmicrosoft_com/EfCJ1nILFYNDp0wER_lYQ-IBa5aTE2vUD-d98rGyeoR1Mw?e=ORXLKe)
谢谢！

ophub, 2024-06-14:
[image]
[image]

你提供的下载链接不可以，我测试了我在线构建的可以。
https://github.com/ophub/amlogic-s9xxx-openwrt/releases

你参考下必备软件包，看看少了什么，看10.11
https://github.com/ophub/amlogic-s9xxx-openwrt/blob/main/documents/README.cn.md

etnperlong, 2024-06-14:
谢谢你的帮忙！我检查过了必备的软件包，发现缺少了 Python 的部分，但我通过 opkg 重新安装后，依然无法通过 Luci 界面更新

我检查了一下在线构建的源，发现源采用的是 master 分支从头编译，而我使用的构建方法则是 使用了 23.05 正式版的 Imagebuilder

通过修改 lua 脚本，我尝试使用 `env` 命令引出了通过 SSH 执行和 LuCI 界面执行的不同

SSH 执行的输出如下
```shell
SHELL=/bin/bash
ENV=/etc/shinit
PWD=/root
LOGNAME=root
HOME=/root
SSH_CONNECTION=192.168.0.2 65432 192.168.0.1 22
TERM=xterm-256color
USER=root
SHLVL=1
PS1=\[\e]0;\u@\h: \w\a\]\u@\h:\w\$ 
SSH_CLIENT=192.168.0.2 65432 22
PATH=/usr/local/sbin:/usr/local/bin:/usr/sbin:/usr/bin:/sbin:/bin
SSH_TTY=/dev/pts/0
_=/usr/bin/env
```

而通过 LuCI 界面执行 env，输出如下
```shell
PWD=/www
SHLVL=1
_=/usr/bin/env
```

两者的重要差别是，LuCI 执行命令时缺少了 PATH 环境变量

于是我在上方我提及的两个脚本的首行添加
```shell
export PATH=/usr/local/sbin:/usr/local/bin:/usr/sbin:/usr/bin:/sbin:/bin
```

再次使用 LuCI 界面操作，即可成功完成更新
[image]

etnperlong, 2024-06-14:
不知道产生差异的原因会不会是 OpenWRT 或 ImmortalWRT 的 LuCI 对于 snapshot 和 正式版本的行为有差异，待我去检查一下他们的代码或查看其他软件包开发者的处理方法，再回来报告

ophub, 2024-06-14:
感谢分享你的发现。

我去imm的luci仓库里搜索了一下，看看是不是有其他app也要显示声明path，但搜了一下没找到有声明的： https://github.com/search?q=repo%3Aimmortalwrt%2Fluci%20export%20PATH%3D&type=code

我上午用你的固件测试更新完整固件时，界面显示lsblk缺失，中断了。然后我就刷了一个我op仓库的试了一下可以。我还发现你的固件有个不同的地方是which总是会返回字符串，以如`which tar`返回`/bin/tar`，`which aaaaa`返回`/bin/……等path变量没有找到aaaaa`，一般来说没有时会返回空，所以 `[[ -n $(which tar) ]]` 检查是可以的，但是你的固件的which检查没有的文件也总是返回非空的字符串，这个shell语法就不对了。

etnperlong, 2024-06-14:
对于我的固件，我应该在构建的时候去除 which 这个依赖，让系统使用 busybox 的 which，即可解决这个问题
但为了更好的兼容性（可能也有部分固件使用了 coreutils ），我的建议是，可以在脚本上添加 PATH 变量引用，以及使用 `$?` 变量来判断 which 执行的返回值，而不是输出字符串，如果返回值大于0则表示依赖不存在
再次感谢你的帮助🙏

ophub, 2024-06-15:
挺好，每消灭一个问题，就向完美又近了一步，感谢分享你的成果。

commit 55bcd69d (2024-06-15): Merge pull request #55 from MyWRT/main

Use busybox which to check dependencies (fix #54)
commit 8bf6b990 (2024-06-15): Use busybox which to check dependencies (fix #54)

Avoid differences in dependency checking behaviour caused by different versions of which.
closed 2024-06-15 by ophub