ophub/luci-app-amlogic #58: 请问N1的这个版本可以使用晶晨宝盒升级 openwrt 吗？

wangjiegulu, 2024-10-20:
19年的时候 N1 设备按照 right.com.cn 中的某个教程安装这个这个版本版本，后来一直没有升级过。
[image]

安装的时候报这个错：
[image]

后台可以展示晶晨宝盒的页面，但是识别不出来 N1 的设备型号：
[image]

按照这个 issue (<https://github.com/ophub/luci-app-amlogic/issues/21>) 手动添加了这个文件：

```
echo "PLATFORM='amlogic'" >/etc/flippy-openwrt-release
```

[image]

想请问下目前可以使用晶晨宝盒安全升级吗？

ophub, 2024-10-21:
可以，在线更新和手动上传更新均可。

wangjiegulu, 2024-10-22:
> 可以，在线更新和手动上传更新均可。

非常感谢！

wangjiegulu, 2024-10-26:
> 可以，在线更新和手动上传更新均可。
你好，在线更新报如下错：
[image]

closed 2024-10-29 by wangjiegulu